Arksine/katapult #168: “Non-hexadecimal number found” error flashing Klipper to EBB36

Heptagon-1, 2025-08-13:
I’m setting up my EBB36 toolhead board via the CANbus guide found [here](https://canbus.esoterical.online/toolhead_flashing#katapult-is-now-installed). Everything is working alright until the last step of flashing Klipper onto the toolhead board, where I’m getting an error that doesn’t seem to make sense. The flashing command is

```python3 ~/katapult/scripts/flashtool.py -i can0 -f ~/klipper/out/klipper.bin -u 17462ab17560```

I have double and triple checked that the UUID is correct and there’s no silly issues like having a space at the end. The error I get is

```
ERROR:root:Flash Tool Error
Traceback (most recent call last):
  File "/home/biqu/katapult/scripts/flashtool.py", line 1057, in main
    sock = CanSocket(args)
  File "/home/biqu/katapult/scripts/flashtool.py", line 583, in __init__
    self._search_canbus_bridge()
  File "/home/biqu/katapult/scripts/flashtool.py", line 758, in _search_canbus_bridge
    det_uuid = convert_usbsn_to_uuid(serial_no)
  File "/home/biqu/katapult/scripts/flashtool.py", line 203, in convert_usbsn_to_uuid
    chip_id = bytes.fromhex(serial_number)
ValueError: non-hexadecimal number found in fromhex() arg at position 5
```

I can’t find any examples of other people having this error across the internet, and I’m pretty stumped as to what the actual issue is. The character at position 5 of the UUID is “a” which last time I checked is definitely a valid hex character. Is anybody able to shed some light on why this might be happening?

I’ve had a look through the flashtool script and I can see there’s a line that logs the value of serial_number
```
serial_no = usb_info["serial_number"]
            logging.info(
                f"Found Klipper USB-CAN bridge on {can_intf}, serial {serial_no}"
            )
```

however I don’t know where this log is saved to and it doesn’t seem to be part of the live output in the terminal.

Arksine, 2025-08-13:
The version of the flashtool you are using appears to be out of date.  The latest version handles canbridge detection errors.

FWIW, that error indicates that the USB Serial Number assigned to the bridge is not the SN automatically generated by the firmware.

Heptagon-1, 2025-08-30:
Updating the most recent version seems to have solved this problem. The same error still occurs but the flashing proceeds successfully afterwards.

closed 2025-08-30 by Heptagon-1